marthoc/GarHAge #8: Unable to publish to broker from other clients

bigalboyo, 2017-11-09:
Hi, 
I've just setup and run this project but now only the ESP8266 can publish to the broker with its open or closed state.  Nothing else can publish anything else to the broker.  Even from the command line on the Pi running the broker, I can't publish anything.  Which also means I can't publish anything to trigger the relay.  Its like the program has taken complete control of the broker.  Any ideas why that might be?

Cheers

marthoc, 2017-11-09:
I can’t see how a client could possibly block other clients from connecting to or publishing to a broker. Can you give me more details about your setup (really just the broker and how it’s set up) so that I can try to replicate and help you debug? 

One thing I’m wondering is if you have specified the same client ID on each device... that would create collisions and constant disconnects/reconnects from and to the broker by the late-joining client that could look like one client was dominating. Client IDs need to be unique.

bigalboyo, 2017-11-13:
Cheers for the response; I realise it is a long shot to put this code as the cause but everything stopped working the moment I ran it for the first time so logically, it got the blame.
As it happens, I ended up having to removed and reinstall mosquitto on the Pi to get the broker running properly again. I've now written my own version of this and everything is going fine. Will chalk it up to experience. cheers

marthoc, 2017-11-13:
Sorry to hear about your experience and that you couldn’t get GarHAge running. Thanks for reporting this, and if you have any more details so that I can examine the code please let me know.

closed 2017-11-13 by marthoc