marthoc/GarHAge #11: Can't compile in Arduino 1.8.5

mattie47, 2018-01-27:
When I try to compile for esp8266, I get:

```
exit status 1
invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
```

config.h:
```
/*   
 * GarHAge
 * a Home-Automation-friendly ESP8266-based MQTT Garage Door Controller
 * Licensed under the MIT License, Copyright (c) 2017 marthoc
 * 
 * User-configurable Parameters 
*/

// Wifi Parameters

#define WIFI_SSID "wifi ssid"
#define WIFI_PASSWORD "password"

// Static IP Parameters

#define STATIC_IP false
#define IP 192,168,1,100
#define GATEWAY 192,168,1,1
#define SUBNET 255,255,255,0

// MQTT Parameters

#define MQTT_BROKER "192.168.1.60"
#define MQTT_CLIENTID "GarHAge"
#define MQTT_USERNAME "asdf"
#define MQTT_PASSWORD "asdf"

// Relay Parameters

#define ACTIVE_HIGH_RELAY true

// Door 1 Parameters

#define DOOR1_ALIAS "Door 1"
#define MQTT_DOOR1_ACTION_TOPIC "garage/door/1/action"
#define MQTT_DOOR1_STATUS_TOPIC "garage/door/1/status"
#define DOOR1_OPEN_PIN D2
#define DOOR1_CLOSE_PIN D2
#define DOOR1_STATUS_PIN D5
#define DOOR1_STATUS_SWITCH_LOGIC "NO"

// Door 2 Parameters

#define DOOR2_ENABLED false
#define DOOR2_ALIAS "Door 2"
#define MQTT_DOOR2_ACTION_TOPIC "garage/door/2/action"
#define MQTT_DOOR2_STATUS_TOPIC "garage/door/2/status"
#define DOOR2_OPEN_PIN D1
#define DOOR2_CLOSE_PIN D1
#define DOOR2_STATUS_PIN D6
#define DOOR2_STATUS_SWITCH_LOGIC "NO"
```

Any idea on what I need to do to get this running? :-)

Kharte55, 2018-01-28:
Did you change anything but the config file e.g the firmware file

marthoc, 2018-01-28:
@mattie47 There must be more debug output that gets printed when the compile fails. I need to know which variable it’s complaining about.

mattie47, 2018-01-28:
@marthoc  I'll try again when I'm home from work re. more debug. 

>I’m also assuming your wifi SSID doesn’t have a space in it in real life?

It does. I did wonder if this may cause a problem..

marthoc, 2018-01-28:
Well the easy way to test is to remove the space and try again to compile. If it works, there’s your problem. I have to double check but I don’t think the ESP8266WiFi library supports SSIDs with spaces.

marthoc, 2018-01-31:
Any update on this @mattie47? I’m planning a new release in the next week or so and if this is actually a bug I’d like to squash it before then.

walthowd, 2018-01-31:
FWIW, it compiles fine for me a with a space in the SSID. Arduino IDE 1.8.5, esp8266 2.4.0

marthoc, 2018-01-31:
I couldn’t replicate either, that’s why I’d like the full compile output.

mattie47, 2018-01-31:
Hey,

Sorry- life’s been busy. I’ll make it a priority today to try again this evening and post all relevant information. 

Thanks,

Matt

mattie47, 2018-02-01:
Please see: https://pastebin.com/WJShtYNC

Removing the space in the ssid doesn't change the output.

marthoc, 2018-02-01:
Can you go into the Library Manager in Arduino IDE and tell me what version of PubSubClient you’re on? The call signatures and methods it’s complaining about are certainly in the version I developed against. Maybe you’re on an old version of PSC?

For an example, take one of the first problems it’s compaining of - that the code is trying to cast const char to const unsigned char in publish(); the current API docs clearly state that both topic and payload are const char and that retained is Boolean: https://pubsubclient.knolleary.net/api.html#publish2.

marthoc, 2018-02-01:
I just checked and I'm on PubSubClient 2.6, which according to GitHub is the latest release.

mattie47, 2018-02-01:
Ahhh - yes! This was my problem. I knew I had the PubSubClient installed, however having also had a look at https://github.com/OpenGarage/OpenGarage-Firmware I didn't realise the only version I had installed was https://github.com/Imroy/pubsubclient which was based on 1.99.1.

Having now also installed PubSubClient 2.6.0, GarHAge is now compiling :-)

Thanks!

closed 2018-02-01 by mattie47

marthoc, 2018-02-01:
Glad to hear it! Enjoy. :)