marthoc/GarHAge #12: OpenHab and publishing status

bradcurtis, 2018-02-01:
Hi,
I have your system working on openhab.  I was wondering what your thoughts were around not publishing status messages every minute or so.  Most of the other openhab implementations continuously publish the status of sensors.  

I am using MQTT.fx.  I can see the status messages during open/close events, but I cannot see them running in a loop

Also, if you want I can write up what I did to get openhab working.  I had to translate some of your messages to work with the switch and contact items

garage/door/1/status
closed

void check_door1_status() {
  int currentStatusValue = digitalRead(door1_statusPin);
  if (currentStatusValue != door1_lastStatusValue) {
    unsigned int currentTime = millis();
    if (currentTime - door1_lastSwitchTime >= debounceTime) {
      publish_door1_status();
      door1_lastStatusValue = currentStatusValue;
      door1_lastSwitchTime = currentTime;
    }
  }
}

marthoc, 2018-02-01:
Yes it was a deliberate choice not to have the door status messages on a repeating loop because it’s not really necessary. The messages are published with the retain flag set, so any time your client (OpenHAB) connects to the broker it will get the latest status. If it’s not changing, then what’s the point of continually publishing it?

bradcurtis, 2018-02-01:
The one thing I noticed is that my nodemcu is not transmitting the messages
all the time. It almost seems like it goes to sleep after long periods

I was just wondering if that was why tasmota and easy esp send status
updates every few minutes

marthoc, 2018-02-02:
Then I think there’s something wrong with your NodeMCU - maybe power supply issue? Or wireless dropping out? Because mine has been running GarHAge in my own garage for months now without a power cycle and it works every time. (That’s not a brag, btw, but it is a fact).

marthoc, 2018-02-02:
And all that being said, it’s trivial to add this feature so I will put it in the next update and make it a configurable option. Sound good?

bradcurtis, 2018-02-02:
Mark,

No worries, I was more curious on the pattern that some people use and why
they do it.

Thanks again for the hard work

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

mrSabic, 2018-03-25:
Hi,
why does it sometimes publish the same message when i close and open the door status pin?

Door 2 open! Publishing to garage/door/2/status...
Door 2 open! Publishing to garage/door/2/status...

gusl79, 2018-10-08:
Hi Mark,

Just wondering if you would be able to share your Openhab setup with Garhage. I've managed to get Garhage working with MQTT.fx but can't seem to get Openahb to trigger my NodeMCU.

I presume it has to do with the translation you had to perform. 

I'd be interested to see how you configured your items