marthoc/GarHAge #20: Incorporate limit switches into relay/motor code

danland6, 2018-04-26:
just wondering how hard it would be for you to modifiy the main ino code so that 

essentially when you send the Open payload
- D2 keeps relay 1 HIGH, until NO limit switch on D5 goes LOW

and when sending closed Paylod

- D1 keeps relay 2 high, untill NO limit switch on D6 goes LOW (with option to maybe over run by a set amount of time,

Essentially i've built a gate that i am trying to automate for my chicken coup, using an old power window regulator/motor for lift, and a gravity latch that after all the tension is removed from the lift cable locks into a slot on one of the side rails, (which is why it would be good to have a slight overrun on close)

I have it all wired up on the bench with 2 relays (for motor reversal), and the 2 limit switches and have the buttons on my home assistant set up for testing, relay up, relay down, and status via one of the limits works fine as expected (i have used your code for my roller door, and front electric gate, and 2 side doors of my shed working great)  

So far from all of my searching and lack of ardunio sketch skills i have not been able to figure out how to make these features work myself yet.

let me know if its something that you'd be keen to do, and if any of my information was not clear enough.

Thanks
Dan

danland6, 2018-04-28:
Hey Mark,
Thanks for the reply, yep makes sense that mostly everyone will not need
this feature for their garage doors!

Awesome that you are keen to help get it working, thanks already, also keen
to see what new features you've added to the existing setup.

And yes you've got it straight, i think the slight offset will be needed on
the close for sure, i will take a few pictures of my gate and post them up
somewhere, unless i can maybe mount one of the magnets to part of the
gravity locking mechanism so that it does not need to over run slightly,
which "may" be possible,

Thanks
Dan

danland6, 2018-04-28:
turns out its as easy as drag and drop, picture quality not amazing due to smashed lens on my phone camera but shows what i'm working with.

[image]
[image]
[image]
[image]
[image]

marthoc, 2018-04-30:
Quick question before I go too far: what microcontroller are you using? NodeMCU? Need to know how many GPIO I’m dealing with.

danland6, 2018-04-30:
Hey there, yep I'm Using the nodeMCU
Cheers
Dan

danland6, 2018-06-16:
Hey, i know its not high on your list but just checking in to see if you had any luck at all?
Cheers
Dan

PaulHam211, 2019-06-02:
This would be great, just what I am looking for too.