marthoc/GarHAge #23: Mqtt issue or code?

codedmind, 2018-09-01:
After some days working well now i'm getting issues when the garage doors is closed the homeassistant is getting wrong status inside a minute it open and close again, the garage door is always close...

If open everything is ok.

Any suggestion for debug?

marthoc, 2018-09-03:
Possible that your esp8266 device is dropping off the wifi network?

sam-wright, 2018-09-03:
Mine does the opposite of this (if open, it alternates between open and close every minute or so). This is because I didn't install a pull-down resistor, so then the door is open, that pin is left floating. Depending on your circuit, you may have left out a pull-up or a pull-down resistor.

codedmind, 2018-09-03:
When I test the first time with version 1 don't have issue only see that
after upload the vldev version.

But in none of the cases I use a pull up resistor :/

marthoc, 2018-09-03:
@sam-wright The pin shouldn’t be floating as I’m enabling the INPUT_PULLUP mode for the door status input pins. So when the door is closed, the pin is pulled low, but when open, the internal pull-up resistor should be pulling it high. Possible though that the resistor is too weak depending on your own environment. @codedmind you can try using a pull-up resistor and see if it solves your problem.

codedmind, 2018-09-04:
Any sechematics to use the pull-up?
Thanks

codedmind, 2018-09-15:
Ok i made the connections an try the both versions (master and dev) in the master the serial connections show normal output but the dev is always publish to mqtt don't now why... same connections , same hardware

`. WiFi connected - IP address: 192.168.1.80
Ready for OTA Upload...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 open! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
[... 485 lines cut ...]
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 closed! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 closed! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 closed! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
Garage Door 3 closed! Publishing to garage/door/3/status...
Attempting MQTT connection...Connected!
Subscribing to GarHAge/api...
Publishing birth message "online" to GarHAge/availability...
Subscribing to garage/door/3/action...
`