marthoc/GarHAge #27: Is there way I could replace Z Wave tilt sensor on garage door instead of using magnetic sensors?

slonkar, 2018-10-09:
I already have Z wave sensor mounted on my garage door. If I am not wrong, you are using magnetic sensors to basically broadcast the state of the door. Is there way I could tweak your code in-order to support Z Wave + relay configuration instead of magnetic switch + relay configuration?

walthowd, 2018-10-09:
Assuming you are using Home Assistant?

I use Z wave sensors and just have the cover set to optimistic and it works fine. 

```
- platform: mqtt
  name: "Garage Door North"
  state_topic: "garage/door/north/status"
  command_topic: "garage/door/north/action"
  availability_topic: "GarHAge/availability"
  optimistic: true
  qos: 0
  retain: false
  payload_open: "OPEN"
  payload_close: "CLOSE"
  payload_stop: "STATE"
  state_open: "open"
  state_closed: "closed"
  payload_available: "online"
  payload_not_available: "offline"
```

Then I have a template switch that I use to mainly control the door, that reports the status via the Z wave sensor:

```
- platform: template
  switches:
    garage_door_north:
      value_template: "{{ is_state('sensor.garage_door_north_tilt_alarm_level', '255') }}"
      turn_on:
        service: cover.open_cover
        data:
          entity_id: cover.garage_door_north
      turn_off:
        service: cover.close_cover
        data:
          entity_id: cover.garage_door_north
      icon_template: >-
        {% if is_state('sensor.garage_door_north_tilt_alarm_level', '255') %}
          mdi:garage-open
        {% else %}
          mdi:garage
        {% endif %}
```

Finally, I also have some Node-RED flows that read the state of the sensors and inject the MQTT messages to keep the state topics on the garage door cover in sync. Not needed in my setup, but just to have everything reporting the same across the board.

marthoc, 2018-10-16:
@slonkar - you can use the template cover component of HA to do this. Essentially what @walthowd showed you except instead of a switch you get a cover device at the end. Useful if you’re using the HomeKit component so that it shows as a Garage Door, for instance. Let me know if you need better instructions...

shoeman22, 2018-12-16:
Just wanted to comment and say that this cover approach worked perfect for me using an [Ecolink 
Z-Wave Plus Tilt Sensor](https://www.amazon.com/Ecolink-Intelligent-Technology-Z-Wave-TILT-ZWAVE2-ECO/dp/B01MRZB0NT?th=1).  

Based on how I did my binary sensors, I had to check for "on" instead of "open".  Here's the relevant config for me.  `binary_sensor.main_bay_garage_door_tilt_binary_sensor` is created automatically by the tilt switch when HA brings it it.  I could probably just use it but I like making explicity binary_sensors so I can use them should I ever decided.

Here's my final config for reference (note I also changed in config.h the mqtt topic)
```
binary_sensor:
  - platform: template
    sensors:
      is_main_bay_garage_door_open:
        value_template: '{% if is_state("binary_sensor.main_bay_garage_door_tilt_binary_sensor", "off") %}false{% else %}true{% endif %}'
        device_class: garage_door
        friendly_name: 'Main Bay Garage Door (tilt)'
        entity_id: binary_sensor.main_bay_garage_door_tilt_binary_sensor

cover:
  - platform: template
    covers:
      main_bay_garage_door:
        friendly_name: "Main Bay Garage Door"
        value_template: "{{ is_state('binary_sensor.is_main_bay_garage_door_open', 'on') }}"
        entity_id: binary_sensor.is_main_bay_garage_door_open
        open_cover:
          service: mqtt.publish
          data:
            topic: garage/main_bay/action
            payload: OPEN
        close_cover:
          service: mqtt.publish
          data:
            topic: garage/main_bay/action
            payload: CLOSE
```

Also, thank you very much @marthoc for garHAge in general.  This is the first time I've ever really mucked around with DIY on a project like this and your instructions and bill of materials were absolutely top notch.