marthoc/GarHAge #28: Relay fires when power is applied

harfordhawk, 2018-11-09:
Mark,
I am trying to use this code very simply with a nodemcu and a 4-relay board.  I power the relay from a separate 5V line from the same usb power source.  Have common grounds between them.  Anyway, everything works great except the power up sequence, which fires the relay. This is not good as you know for when I have a power outage, the garage will open when power is restored.
I have tried setting active high and active low.  Do you have any ideas how I can stop the triggering on power up?  Thanks

Mike1082, 2018-11-12:
I think this has something to do with certain pins on the NodeMCU being pulled high or low during boot. I am running into this exact issue myself with a NodeMCU and 2 relay module while automating my doorbell (essentially the same thing as this project). Are you using D0 (GPIO 16)? [See here for more information](https://github.com/thehookup/Wireless_MQTT_Doorbell/blob/master/GPIO_Limitations_ESP8266_NodeMCU.jpg).

harfordhawk, 2018-11-12:
have not tried D0, but tried many of the other 'safe' pins, and it still
fires on power up.  Ofcourse , we can't have that for a garage door as if
we lose power, say while we are away, the garage door would open

Mike1082, 2018-11-13:
Ok I was just able to get around this on my doorbell sketch. I don't know if this is a preferred method, or if there is a better way, but my relays are no longer firing on bootup (using D0 and D2 on NodeMCU). I did a digitalWrite to HIGH before setting the pinMode in the void setup() function. I found the solution [here](https://electronics.stackexchange.com/questions/231742/how-to-enable-output-pin-to-high-by-default-in-arduino-uno).

```
digitalWrite (RELAY1, HIGH);
digitalWrite (RELAY2, HIGH;
pinMode (RELAY1, OUTPUT);
pinMode (RELAY2, OUTPUT);
```

If this is bad, and someone knows a better way, please let me know.