marthoc/GarHAge #32: Preventing Multiple Triggers

DenseComet, 2019-02-10:
If I click open in home assistant, and then hit open again when the garage is opening, the garage stalls because another open signal has been sent to it. What would be the best way to ignore a second trigger within a 15 or so second period after the first trigger? There is no clean way to do this in home assistant and doing it within GarHAge would be more platform agnostic.

I'm not sure if this solution is optimal, but one method would be setting a variable when the door is triggered. A loop can detect a change and then reset it after 15 seconds. When triggering the door, this variable can be checked to make sure it is in the default state. Essentially, after triggering the door once, disabling any subsequent triggers for 15 seconds.

heymoe, 2019-02-14:
I would think multiple triggers would be the expected behavior for a garage door. If you locked out the buttons for 15 seconds then there would be no way to stop the door in any position other than fully open or fully closed. 

That being said, I could see you wrapping the call to the triggerDoorAction function inside the callback function with the same if logic that is used the the check_door#_status function (debounce) that prevents false activation readings of the reed switch while the door is moving. 

Like So:
```
  unsigned int currentTime = millis();
  if (currentTime - lastClickTime >= lockOutTime) {
    triggerDoorAction(topicToProcess, payloadToProcess);
    lastClickTime = currentTime;
  }
```
You will need to define the lastClickTime and lockOutTime vars as a global just like door1_lastSwitchTime.

```
unsigned long lastClickTime = 0;
int lockOutTime = 15;
```
Defining lockOutTime = 0 would maintain the current behavior of allowing multiple clicks.

DenseComet, 2019-02-14:
I'll try this out when I get a chance thanks!

I would like to make sure that the door is not in any state but fully open or fully closed. This is because I have no way to detect if it is in an in between state. If it gets stuck in an in between state, home assistant would still show that it is closed. If it can be in an in between position, I can never have a guarantee that it is completely closed.

heymoe, 2019-02-14:
Ahhh, I worked around your in between state by having it so that when the reed switch is activated (closed) that means my garage door is fully closed instead of it meaning fully open.

marthoc, 2019-02-16:
@DenseComet - you should install the reed switch so that the magnets are close together when the door is completely closed. That way, any other position will cause Home Assistant to report the door as open.

aditsachde, 2019-06-08:
@heymoe's solution worked perfectly. Obviously putting the reed switch in a better location would be optimal, but for my usecase, it works perfectly. 

https://github.com/aditsachde/GarHAge/blob/master/GarHAge/GarHAge.ino#L105

closed 2019-06-08 by aditsachde