marthoc/GarHAge #33: action triggered on connection to relay

ehudkaldor, 2019-10-28:
not sure if this is something you can help with, but here goes:

as soon as i connect the relay to the garage door trigger (COM and NO), it triggers an action. at this point, i can control the garage door via MQTT, but the physical button stops working. i am not sure if i'm connecting something wrong, but this happens even if the relay is not corrected to power. and i've tried it with 2 different relays.

help. please.

mrSabic, 2019-10-28:
Strange, normally this would happen if you connect the wires to COM and NC, maybe the relay is broken. Test it with a multimeter, there shouldn't be continuity between COM and NO (when the relay is not powered).

cmcapone, 2020-05-31:
I had this same issue.  My relay was active-low (which I didn't realize) and I hadn't set ACTIVE_HIGH_RELAY to false.  I updated the .h, uploaded the new sketch and it worked perfectly (physical buttons work again).

cben0ist, 2021-05-03:
> I had this same issue. My relay was active-low (which I didn't realize) and I hadn't set ACTIVE_HIGH_RELAY to false. I updated the .h, uploaded the new sketch and it worked perfectly (physical buttons work again).

That just saved my day. I was in the same case using an ESP-01/01S
And here is how to set ACTIVE_HIGH_RELAY to false in ESPHome: https://esphome.io/components/switch/gpio.html#active-low-switch

Thank you!