marthoc/GarHAge #34: Garage periodically appears as unavailable in HA

nadnerb33, 2019-12-30:
I've set up a ESP8266 with a reed switch to simply monitor the status of the garage door (but not control it).
It seems to work well and the status in HA is responsive when I open/close the garage door.
The problem is several times per hour it shows the status as "unavailable" for what I assume only a split second before providing a "closed" status again. During this time the door is closed and power supply is fine. The constant switching between "unavailable" and "closed" just clogs up the HA logs and makes it look like the garage was recently open even if it's been shut for hours.

Things I've tried:
- different USB power supplies plugged into mains
- USB battery pack
- placing device right next to wifi router
- upgraded my Arduino board manager for esp8266 from 2.4 to 2.6.3
- added `#define MQTT_KEEPALIVE 10` to match mosquitto keepalive

So I don't think it is an issue with wifi signal or power.
In the Mosquitto log I see:

```
1577719913: Socket error on client GarHAge, disconnecting.
1577719914: New connection from 192.168.0.153 on port 1883.
1577719914: New client connected from 192.168.0.153 as GarHAge (c1, k15, u'').
1577720530: Socket error on client GarHAge, disconnecting.
1577720530: New connection from 192.168.0.153 on port 1883.
1577720530: New client connected from 192.168.0.153 as GarHAge (c1, k15, u'').
```
It often disconnects and reconnects within a second so am wondering if the `keepalive` values in the client or broker need to be adjusted?

Thanks

cmcapone, 2020-07-03:
Curious if you ever found a solution? I'm having the same issue and confirmed that the actual device has stayed connected to my router for ~100 hours.
Note that occasionally I see a longer delay in addition to the "instant" disconnect/reconnects - strangely almost always in the 14-16 minute range.

rumlak, 2020-07-03:
I am running the dev version and have never had a problem.  Been running for a couple of years with zero intervention.

cmcapone, 2020-07-03:
Interesting, I was simply running the "master" branch.  I will try the dev version - a bit painful to go and retrieve the device, but it looks like the dev version has OTA updates (master does not).  It is worth a try, thank you - I'll report back if there are any results.

nadnerb33, 2020-07-06:
I don't think I ever found a proper solution/cause of the problem.
 
After testing various things I decided to just leave it as is and wire it all up in the garage.
I think I made a tweak in Home Assistant so it didn't show the millisecond `unavailable` statuses.
Looking at my config, I assume I edited the `cover` component:

    payload_not_available: "offline_nope"

I guess that should be "offline", but changing it to any over value prevents it from reporting as `unavailable`.
Bit hacky, and not sure what happens when it's truely unavailble but it seemed to solve my issue.

cmcapone, 2020-07-07:
Thanks for getting back to me with your results.  
For what it is worth, I did switch to the "dev" branch and made a tweak to it and have had much better (but not perfect) results.

I added
`client.setKeepAlive(60);`

to both void setup() and the reconnect() functions.  

I still occasionally get the instantaneous "unavailable" but much less frequently.  Unfortunately it seems that a couple are still in the multiple minute range (consistently 10-15 minutes).  When this occurs, you cannot command the doors to change state.