marthoc/GarHAge #35: Adding Pedestrian doors

davidm2232, 2020-06-22:
It would be nice to have the option to add additional inputs for door sensors for pedestrian doors. Would allow the ability to control auto open/close of doors from 
HomeAssistant. I tried it on my own but am not getting the statuses to update. 
[GarHage-4_doors.zip](https://github.com/marthoc/GarHAge/files/4816081/GarHage-4_doors.zip)

davidm2232, 2020-06-22:
And the HA configuration.yaml:

binary_sensor:
  - platform: mqtt
    name: "Garage Door 1"
    state_topic: "garage/door/1/status"
    payload_on: "open"
    payload_off: "closed"
    availability_topic: "GarHAge/availability"
    device_class: opening
    qos: 0

  - platform: mqtt
    name: "Garage Door 2"
    state_topic: "garage/door/2/status"
    payload_on: "open"
    payload_off: "closed"
    availability_topic: "GarHAge/availability"
    device_class: opening
    qos: 0  

  - platform: mqtt
    name: "Garage Door 3"
    state_topic: "garage/door/3/status"
    payload_on: "open"
    payload_off: "closed"
    availability_topic: "GarHAge/availability"
    device_class: opening
    qos: 0

  - platform: mqtt
    name: "Garage Door 4"
    state_topic: "garage/door/4/status"
    payload_on: "open"
    payload_off: "closed"
    availability_topic: "GarHAge/availability"
    device_class: opening
    qos: 0

rumlak, 2020-06-23:
I am pretty sure this is available in the dev version.

davidm2232, 2020-06-23:
I just looked and it would appear they are. I'll upload them tonight and see if it works.