marthoc/GarHAge #38: Gate Status Icons

jy1919, 2020-11-04:
I have used this with HASSIO for my sliding gate opener, and it works pretty well. But i would like to use an open and a close icons for the gate status.
[image] 
[image]

what I am getting is this:
[image]

Is there a way to accomplish this? Thanks for your help.

splasheye, 2021-03-04:
Hi I would like to know the same thing. Is it anyway we could change the icon used in home assistant?

alertedsnake, 2021-03-04:
That's a HomeAssistant issue, use the "customize" feature.